PR2/pr2_mechanism #330: Incompatible with ROS Kinetic

yswhynot, 2017-05-25:
I've been trying to build from source on Ubuntu 16.04 with ROS Kinetic and I've ran into the following compilation error:
```
In file included from /opt/ros/kinetic/include/hardware_interface/robot_hw.h:35:0,
                 from /opt/ros/kinetic/include/controller_interface/controller_base.h:37,
                 from /opt/ros/kinetic/include/controller_interface/controller.h:35,
                 from /home/yisha/workspace/gsoc/catkin_ws/src/pr2/pr2_mechanism/pr2_controller_interface/include/pr2_controller_interface/controller.h:48,
                 from /home/yisha/workspace/gsoc/catkin_ws/src/pr2/pr2_mechanism/pr2_controller_manager/test/controllers/test_controller.h:1,
                 from /home/yisha/workspace/gsoc/catkin_ws/src/pr2/pr2_mechanism/pr2_controller_manager/test/controllers/test_controller.cpp:1:
/opt/ros/kinetic/include/hardware_interface/internal/interface_manager.h: In instantiation of ‘T* hardware_interface::InterfaceManager::get() [with T = pr2_mechanism_model::RobotState]’:
/opt/ros/kinetic/include/controller_interface/controller.h:110:30:   required from ‘bool controller_interface::Controller<T>::initRequest(hardware_interface::RobotHW*, ros::NodeHandle&, ros::NodeHandle&, controller_interface::ControllerBase::ClaimedResources&) [with T = pr2_mechanism_model::RobotState; controller_interface::ControllerBase::ClaimedResources = std::vector<hardware_interface::InterfaceResources>]’
/home/yisha/workspace/gsoc/catkin_ws/src/pr2/pr2_mechanism/pr2_controller_manager/test/controllers/test_controller.cpp:131:61:   required from here
/opt/ros/kinetic/include/hardware_interface/internal/interface_manager.h:194:21: error: no matching function for call to ‘pr2_mechanism_model::RobotState::RobotState()’
         iface_combo = new T;
                     ^
In file included from /home/yisha/workspace/gsoc/catkin_ws/src/pr2/pr2_mechanism/pr2_controller_interface/include/pr2_controller_interface/controller.h:46:0,
                 from /home/yisha/workspace/gsoc/catkin_ws/src/pr2/pr2_mechanism/pr2_controller_manager/test/controllers/test_controller.h:1,
                 from /home/yisha/workspace/gsoc/catkin_ws/src/pr2/pr2_mechanism/pr2_controller_manager/test/controllers/test_controller.cpp:1:
/home/yisha/workspace/gsoc/catkin_ws/src/pr2/pr2_mechanism/pr2_mechanism_model/include/pr2_mechanism_model/robot.h:130:3: note: candidate: pr2_mechanism_model::RobotState::RobotState(pr2_mechanism_model::Robot*)
   RobotState(Robot *model);
   ^
/home/yisha/workspace/gsoc/catkin_ws/src/pr2/pr2_mechanism/pr2_mechanism_model/include/pr2_mechanism_model/robot.h:130:3: note:   candidate expects 1 argument, 0 provided
/home/yisha/workspace/gsoc/catkin_ws/src/pr2/pr2_mechanism/pr2_mechanism_model/include/pr2_mechanism_model/robot.h:126:7: note: candidate: pr2_mechanism_model::RobotState::RobotState(const pr2_mechanism_model::RobotState&)
 class RobotState : public hardware_interface::HardwareInterface
       ^
/home/yisha/workspace/gsoc/catkin_ws/src/pr2/pr2_mechanism/pr2_mechanism_model/include/pr2_mechanism_model/robot.h:126:7: note:   candidate expects 1 argument, 0 provided
pr2/pr2_mechanism/pr2_controller_manager/CMakeFiles/controller_test.dir/build.make:62: recipe for target 'pr2/pr2_mechanism/pr2_controller_manager/CMakeFiles/controller_test.dir/test/controllers/test_controller.cpp.o' failed
```
The interface_manager of Kinetic requires empty constructor but the RobotState does not have one. May I know will it work without problem if I create an empty constructor for RobotState?

k-okada, 2017-10-21:
@TheClearpathDash can you create kinetic-devel branch ? so that we can push our fix https://github.com/k-okada/pr2_mechanism/tree/kinetic-devel

TheDash, 2017-10-21:
@k-okada I no longer am paid to be responsible for the PR2.

ipa-mdl, 2017-10-21:
@k-okada: please test https://github.com/ros-controls/ros_control/pull/285

k-okada, 2018-02-13:
@ipa-mdl sorry for late, your ros-controls/ros_control#285 works fine for us!