PR2/pr2_mechanism #337: pr2_mechanism_model doesn't compile on Mac OS X

jna29, 2018-06-12:
When trying to compile `pr2_mechanism_model`, I get the error:
`/Users/joshuaaduol/catkin_ws/src/pr2_mechanism/pr2_mechanism_model/include/pr2_mechanism_model/joint.h:81:3: error: use of undeclared identifier 'boost'`
`boost::shared_ptr<const urdf::Joint> joint_;`. 

This doesn't seem to be a problem in my Ubuntu installation where the same `catkin_make -j1 -l1 -DCATKIN_WHITELIST_PACKAGES="pr2_mechanism_model"` compiles correctly.

The problem does go away when I include `<boost/shared_ptr.hpp>` but I get this next error which oddly enough manages to find my `boost` installation (I am using `boost 1.67.0`):

`/Users/joshuaaduol/catkin_ws/src/pr2_mechanism/pr2_mechanism_model/src/robot.cpp:179:35: error: no viable overloaded '='
      joint_states_[js_id].joint_ = model_->robot_model_.getJoint(t->joint_names_[j]);
      ~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:547:18: note: candidate function not viable: no known conversion from 'urdf::JointConstSharedPtr' (aka 'shared_ptr<const urdf::Joint>') to
      'const boost::shared_ptr<const urdf::Joint>' for 1st argument
    shared_ptr & operator=( shared_ptr const & r ) BOOST_SP_NOEXCEPT
                 ^
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:658:18: note: candidate function not viable: no known conversion from 'urdf::JointConstSharedPtr' (aka 'shared_ptr<const urdf::Joint>') to
      'boost::shared_ptr<const urdf::Joint>' for 1st argument
    shared_ptr & operator=( shared_ptr && r ) BOOST_SP_NOEXCEPT
                 ^
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:683:18: note: candidate function not viable: no known conversion from 'urdf::JointConstSharedPtr' (aka 'shared_ptr<const urdf::Joint>') to
      'boost::detail::sp_nullptr_t' (aka 'nullptr_t') for 1st argument
    shared_ptr & operator=( boost::detail::sp_nullptr_t ) BOOST_SP_NOEXCEPT
                 ^
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:556:18: note: candidate template ignored: could not match 'boost::shared_ptr' against 'std::__1::shared_ptr'
    shared_ptr & operator=(shared_ptr<Y> const & r) BOOST_SP_NOEXCEPT
                 ^
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:567:18: note: candidate template ignored: could not match 'auto_ptr' against 'shared_ptr'
    shared_ptr & operator=( std::auto_ptr<Y> & r )
                 ^
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:576:18: note: candidate template ignored: could not match 'auto_ptr' against 'shared_ptr'
    shared_ptr & operator=( std::auto_ptr<Y> && r )
                 ^
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:598:18: note: candidate template ignored: could not match 'unique_ptr' against 'shared_ptr'
    shared_ptr & operator=( std::unique_ptr<Y, D> && r )
                 ^
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:607:18: note: candidate template ignored: could not match 'unique_ptr' against 'shared_ptr'
    shared_ptr & operator=( boost::movelib::unique_ptr<Y, D> r )
                 ^
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:665:18: note: candidate template ignored: could not match 'boost::shared_ptr' against 'std::__1::shared_ptr'
    shared_ptr & operator=( shared_ptr<Y> && r ) BOOST_SP_NOEXCEPT`

Any help with this would be greatly appreciated.

v4hn, 2018-06-12:
The problem is that you apparently use a more current version of urdfdom on your OS X than on your ubuntu.
They changed to `std::shared_ptr` at some point and in theory the pr2 packages should use `urdf::*SharedPtr` for compatibility instead of `boost::shared_ptr<*>`.
However, nobody did this yet. Feel free to submit a patch.

This problem breaks the build in everything past kinetic on Ubuntu too.

jna29, 2018-06-14:
This seemed to solve the problem. Thanks.

v4hn, 2018-06-14:
Where's the pull-request then? ;)